spooknik/FLAP #1: Feature request: Set fixed temp after loading

appleimperio, 2024-07-29:
One of the new things that Prusa does when loading the filament and the printer is not printing. When it finish loading the filament it lowers the temperature usually 175. It would be awesome if you can create a variable to set the temp the user want after the filament is loaded. Other things you might update the instructions in the installation part that the user needs tu add the flap_main.cfg to their printer.cfg in the form of [include flap_main.cfg] I have it working but for newbies. For those that don't have an unload button just uncomment this lines in the flap_main.cfg
````
#Filament Unload macro
#[gcode_button filament_unload] LER disablesd no unload button
#pin: ^!gpio0 # <- Assign your pin here
#release_gcode:  # filament unload procedure   
#  filament_unload_init
#press_gcode: # do not add any macro call here
````
Thanks I really was waiting for something like this.

spooknik, 2024-07-30:
Hello! Thanks for the suggestion, I will be writing more documentation in the next few days.

Regarding reducing the temp of the hotend after loading, good idea. I'll add it soon.

spooknik, 2024-07-31:
I added this feature in V1.2. Check the variables: 

```
variable_cooldown_load          :True 
variable_cooldown_load_temp     :185 
```

closed 2024-07-31 by spooknik