spooknik/FLAP #5: Unable to parse '' as a literal: invalid syntax (, line 0)

AndyL069, 2024-09-26:
[image]

incur, 2025-03-21:
I had exactly the same error. I was able to solve it by manually removing the filament and reloading it with FLAP. 
I suspect that it is because the filament information is stored during loading, which is not known during the initial setup. 
For a workaround during the first setup you could perhaps call the LOAD_PROMPT directly so that you can select the loaded filament.

PaPashka, 2025-04-18:
> I had exactly the same error. I was able to solve it by manually removing the filament and reloading it with FLAP.
> I suspect that it is because the filament information is stored during loading, [...]

That's right. I had the same problem. It's all because the necessary variables were not saved yet during the first attempt to unload the filament. It is necessary to somehow check that the information about the currently loaded filament exists. I am not yet strong in the features of the klipper macros and have not yet found the time to figure this out. Therefore, as a quick solution, I also first initiated the initial loading via a macro. But perhaps the author already knows how to implement the necessary check.

Ithriveontacos, 2025-05-29:
A check to see if the variable is empty and then if it is, bring up the same filament loading prompt so you can just select what's loaded. I've got no clue how to write it but it sounds like a great fix.