spooknik/FLAP #10: Filament sensor assignment problem

laptop-laptop, 2025-04-13:
Hello, I am using a Prusa i3MK3S+. The filament sensor is connected to an Einsy Rambo board at PK0. 

As mentioned in Step 4 of the Installation, I assigned the filament sensor using 

`switch_pin: !PK0`

in flap_main.cfp and commented out the section for "button for unloading". 

Klipper gave an error:

`pin PK0 used multiple times in config`

I went to /config/klipper-prusa-mk3s/mk3s to comment out the `switch_pin: !PK0` in einsy-rambo.cfg but then it gave another error:

`Option 'switch_pin' in section 'filament_switch_sensor fsensor' must be specified`

I got the einsy-rambo.cfg file by following the steps in: https://github.com/dz0ny/klipper-prusa-mk3s

Please help.

spooknik, 2025-04-15:
> pin PK0 used multiple times in config

Check your klipper config for `PK0 ` because it is being used more than once.

laptop-laptop, 2025-04-19:
Thank you. What part of config should I comment out? As mentioned, I found another mention of PK0 in  /config/klipper-prusa-mk3s/mk3s/einsy-rambo.cfg but when I commented it out, I got other errors.

Is there a way that I can use your codes but have the `switch_pin: !PK0 ` commented out and defined somewhere else like /config/klipper-prusa-mk3s/mk3s/einsy-rambo.cfg in my case?